zvldz/vacuum #8: New room navigation beta firmware on S50/S55

tiagofreire-pt, 2019-12-25:
There is a new available beta firmware that will allow to test the funccionalities of room navigation and others, similary the ones present on the new S60 model.

In: https://www.facebook.com/groups/1963580733664463/permalink/2876867562335771/

Could we easily determine the URL to download it?

Thanks.

zvldz, 2019-12-25:
No information yet. It's a closed test.

tiagofreire-pt, 2019-12-25:
It seems to the 3.5.7_002008. 

It has 60 MB, may be an OTA/parcial update?

zvldz, 2019-12-25:
Yeah, it's probably a partial update. Since the firmware usually takes ~106MB.

tiagofreire-pt, 2019-12-25:
Could we guess the link and start to decompile it?

zvldz, 2019-12-25:
As I understand it now, firmware is distributed to beta testers by uid (mi account).
We should wait for the open beta test to start.

zvldz, 2019-12-31:
https://vacuumz.info/download/gen2/stock/v11_002008.fullos.fd043420-6ddb-4e54-bdb7-a8deec19f0fd.pkg

tiagofreire-pt, 2019-12-31:
> https://vacuumz.info/download/gen2/stock/v11_002008.fullos.fd043420-6ddb-4e54-bdb7-a8deec19f0fd.pkg

This is awesome!!!

Is it rootable?

zvldz, 2019-12-31:
So far there is no root version, a lot has changed in the firmware.

MaluNoPeleke, 2020-01-06:
Is the Valetudo RE version safe to install (= still easy to flash with future updates etc.)?

zvldz, 2020-01-06:
Upgrade firmware only by resetting to factory settings. I haven't solved this problem yet.

MaluNoPeleke, 2020-01-06:
So, as I have already installed the 1910 Valetudo RE firmware I have to factory reset it and then flash the 2008 build to upgrade!?

zvldz, 2020-01-06:
Yes. But before that I was replacing the miio-client and it allowed update the firmware without resetting it. It doesn't work yet.

zvldz, 2020-01-15:
Closed #8

tiagofreire-pt, 2020-01-15:
Just to add, this firmware is being distributed across the users on the Chinese server, independently the beta testing program announced a few weeks ago by the Roborock company.

zvldz, 2020-01-15:
The beta test was about a week, then they started giving it to everyone.

holypetrus, 2020-01-17:
The new firmware is out now. Is it possible to use it with valeteduo?

zvldz, 2020-01-21:
Yeah, I use 2008 firmware with valetudo re.

hal295, 2020-01-22:
Hi there, I updated yesterday robot to new 2008 firmware throught Valetudo RE firmware update tool. 
Update was successfull, but I lost Valetudo and now I cant downgrade firmware or install Valetudo.
Any suggestions please? I tried also XVacuum Firmware and FloleVac App.

zvldz, 2020-01-22:
You can try it via mirobo:
mirobo --ip=192.168.1.xx --token=xyz raw-command miIO.ota '{"mode":"normal", "install":"1","app_url":"https://vacuumz.info/download/gen2/vacuum_valetudo_re_2008_beta2.pkg", "file_md5":"96d1540f32a3fc910dbb037efb07d53e","proc":"dnld install"}'
Sending cmd miIO.ota with params {'mode': 'normal', 'install': '1', 'app_url': 'https://vacuumz.info/download/gen2/vacuum_valetudo_re_2008_beta2.pkg', 'file_md5': '96d1540f32a3fc910dbb037efb07d53e', 'proc': 'dnld install'}

holypetrus, 2020-01-22:
Reset robot as described here: https://github.com/dgiese/dustcloud/wiki/Xiaomi-Vacuum-Robots-Factory-Reset. After that you can start over again and install firmware with Valetudo. Beware, if you want to do a firmware update in future you have to do a firmware reset beforehand. Thats because it does not support Updates via Lan.

fredless, 2020-01-26:
> Yeah, it's probably a partial update. Since the firmware usually takes ~106MB.

Looks instead like the 2008 firmware is a full disk image as in the past, rather they trimmed out a lot of unnecessary stuff (some of the more prominent things include X11, perl libraries, package management, terminfo, **openssh**).

tiagofreire-pt, 2020-03-05:
It's possible to revert to a previous version after installing the 2008 and onwards firmware?

holypetrus, 2020-03-05:
> Is it possible to revert to a previous version after installing the 2008 and onwards firmware?

As Far as I know the image which is stored on the backup partition is relevant. Since normally it won’t be overwritten you can revert always to the factory state like when you received the robot. Hope I’m right on this one..

t773, 2020-04-19:
> Looks instead like the 2008 firmware is a full disk image as in the past, rather they trimmed out a lot of unnecessary stuff (some of the more prominent things include X11, perl libraries, package [...]

So the 2008 firmware is not rootable then? Is there still way to update with  openssh put back in? I am very new to this robot and am still learning.

zvldz, 2020-05-31:
closed #8

closed 2020-05-31 by zvldz