zvldz/vacuum #9: Reboot workaround

MaluNoPeleke, 2020-01-16:
To avoid the nightly reboot which might result in a stock firmware reset you can add the following code:

```
# nano /etc/crontab

0 23    * * *   root    service rrwatchdoge stop
10 23   * * *   root    rm /mnt/data/rockrobo/rrlog/*REL -R
15 23   * * *   root    /root/check_all
0 4     * * *   root    service rrwatchdoge start

# nano /root/check_all

umount /mnt/updbuf
umount /mnt/data
umount /mnt/reserve
umount /mnt/default
fsck -y /dev/mmcblk0p10
fsck -y /dev/mmcblk0p1
fsck -y /dev/mmcblk0p11
fsck -y /dev/mmcblk0p6
mount /dev/mmcblk0p10 /mnt/updbuf
mount /dev/mmcblk0p1 /mnt/data
mount /dev/mmcblk0p11 /mnt/reserve
mount -o ro /dev/mmcblk0p6 /mnt/default

# service cron restart
```

zvldz, 2020-01-16:
Thank you, I've seen it before. I don't see a problem with the night reboot. And I haven't had a factory reset, although I don't know what it has to do with it. The 2008 firmware doesn't have the classic cron and fsck. But I will try to add fsck to the firmware.

MaluNoPeleke, 2020-01-22:
Here is the full version: https://github.com/MadJoker0815/roborock_nologs

zvldz, 2020-01-22:
It doesn't work on 2008 firmware.
Needs an adaptation.
The firmware 2008 has no crontab. It contains cron from busybox. That's the first thing I saw.

zvldz, 2020-01-22:
'service' is absent too.

pidator, 2020-02-11:
> The firmware 2008 has no crontab. It contains cron from busybox.

> 'service' is absent too.

Is it the same for Gen1 4004 firmware? I didn't found information about the changes/differences from 3468 to 4004... (except the difference miio client 3.3.9 vs 3.5.4)

zvldz, 2020-02-12:
In the firmware 4004 full ubuntu.
The shortened version is still only in the firmware 2008.

closed 2020-04-29 by zvldz