zvldz/vacuum #10: Question: what is recommended version for GEN 1?

Space2Man, 2020-01-24:
Hi,

on valetudo page I see a warning to not use firmware 4004 but I see prebuilt images with this firmware version. Furthermore it seems like 3600 is older then e.g. 3532?

```
v11_003532.fullos.pkg | 2019041901REL | 3.5.0_003532
v11_003570.fullos.pkg | 2018120302REL | 3.3.9_003570
v11_003600.pkg | 2018112300REL | 3.3.9_003600
v11_004004.amhd98763.fullos.pkg | 2019090500REL | 3.5.4_004004
```

Currently I am running 3600 with valetudo-RE. Just wondering :)

But a big thanks from my side to all the people creating this solution ...

zvldz, 2020-01-26:
I recommend firmware 3532. Firmware 4004 had problems with valetudo. But I don't have the first generation to check if valetudo has changed in the firmware 4004.

kockockockoc, 2020-02-10:
I use 4004 build (one of old prebuild with valetudo_re 0.4.0-RE7) without any issues. GEN1

mvmaastricht, 2020-02-28:
> I use 4004 build (one of the old prebuild with valetudo_re 0.4.0-RE7) without any issues. GEN1

Have you also tested 4004 with Valetudo RE 0.9.0? That a great update!

closed 2020-03-05 by zvldz