zvldz/vacuum #14: Valetudo install on 2008

dimmuboy, 2020-02-23:
Is there any other way to install Valetudo with _vacuum_2008_beta4_ because my standard way to update firmware doesn't work.

After install of _vacuum_2008_beta4.pkg_ I can't update any other firmware (e.g. Valetudo).
**miio** inform me
`ERROR:miio.updater:No request was made..`
**rrcc** give me a error:
```
Robot does not send firmware package request!
Make sure your firewall accepts incoming tcp connections for 192.168.6.17 -> 192.168.6.21:8080 and try again...
```

Moreover downgrade does not work either

zvldz, 2020-02-24:
OTA in new firmware is limited.
Need to update via Valetudo
Settings/Info -> Firmware update

dimmuboy, 2020-02-24:
Thank you so much. Your solution from Telegram group help me.
Upgrade with OTA command
`mirobo --ip=192.168.1.xx --token=xyz raw-command miIO.ota "{'mode':'normal', 'install':'1','app_url':'https://vacuumz.info/download/gen2/vacuum_valetudo_re_2008.pkg', 'file_md5':'cec5d20afb189e119b164ceb5324d058','proc':'dnld install'}"`

closed 2020-02-24 by dimmuboy