zvldz/vacuum #15: Q: How do you bypass the region locking?

richardracko, 2020-03-04:
Hi,

I own 2nd genenration of Roborock (S55). It was purchased from the EU stock and therefore it was locked to EU region.

I've flashed your pre-built FW: vacuum_2prc_1910.pkg and successfully get rid of the lock. I was able to connect vacuum to China mainland servers. However, I wanted to build my own FW and therefore I've done a factory reset of the device. After that I was not able to use the vacuum with EU nor China servers. **MiHome app was saying that device is locked to China region on both servers!** Yes even if I was connected to China Mainland!

I ended up with flashing the vacuum_2prc_1910 FW back to device. I would be very grateful if you can advice how is the region lock bypass done in your prebuilt FW and how can I patch it myself.

Thanks!

zvldz, 2020-03-05:
If the region has been changed, it will remain even after resetting to the factory settings.
To bypass the lock, a patch is used that disables time zone check in AppProxy.
https://github.com/zvldz/vacuum/blob/master/custom-script/custom_appproxy_patcher.sh

richardracko, 2020-03-05:
Thanks for clarification. You are doing great job with this FW customization!

closed 2020-03-05 by richardracko