zvldz/vacuum #19: New voice packages for firmware 2008

MaluNoPeleke, 2020-03-07:
I recently updated the firmware of my S50 to your latest 2008 version.
Thanks a lot for the great work! 
After some time I experienced some English voice information coming from the device in special cases even if I use a German voice package. 
That is why I wonder if the voice packages need to be updated as well (most of the voice information coming from the device are correctly played in German)? 
If yes, could you please upload them to vacuumz.info?

zvldz, 2020-03-09:
Please try this package - https://cnbj2.fds.api.xiaomi.com/tanosv/app/voice-pkg/package/de.pkg

zvldz, 2020-03-09:
Try it https://vacuumz.info/download/voice/s5/2008/de.pkg

Thank Sebastian.

closed 2020-03-12 by zvldz