zvldz/vacuum #26: Allow local updates on gen2/v2008

JohnRev, 2020-05-11:
Gen2 vacuum on FW v2008 with miio client 3.3.9 does not allow local (http) OTA firmware updates.
OTA updates using https links with valid certificates are working with local OTA update.

This patches the check in AppProxy to allow LAN updates on HTTP for v2008 firmware with miio, so that the scripts like `flasher.py` can still work:

```
sed -i 's/\x7F\xF4\x29\xAF\x13\x79\x03\xF0\xDF\x03\x53\x2B\x7F\xF4\xCD\xAC\x21\xE7\x38\x46\xF0\xF7\xE2\xEB/\x7F\xF4\x29\xAF\xD3\x78\x03\xF0\xDF\x03\x50\x2B\x7F\xF4\xCD\xAC\x21\xE7\x38\x46\xF0\xF7\xE2\xEB/g' AppProxy
```

Pattern:

```
Original Hex:  7F F4 29 AF 13 79 03 F0 DF 03 53 2B 7F F4 CD AC 21 E7 38 46 F0 F7 E2 EB
Patched Hex: 7F F4 29 AF D3 78 03 F0 DF 03 50 2B 7F F4 CD AC 21 E7 38 46 F0 F7 E2 EB
```

md5sum of patched AppProxy for v2008 is `fb89529e947607a8cf7dd965dcd2f057`

zvldz, 2020-05-12:
If it really works, it's very cool. Thanks.

zvldz, 2020-05-12:
While I got an update error

`
2020.5.12 6:46:14 TID 10158 pThreadMessageListener:597 IPC message listener thread (10158) started   
2020.5.12 6:46:14 TID 10152 main:338 Opening package info file /mnt/data/.temp/Update.pkg.inf        
2020.5.12 6:46:14 TID 10158 pThreadMessageListener:604 Waiting for messages...                       
2020.5.12 6:46:14 TID 10159 pSnThreadWorker:153 Status notifier worker thread (10159) started        
2020.5.12 6:46:14 TID 10152 main:344 Deleting update package info file /mnt/data/.temp/Update.pkg.inf
2020.5.12 6:46:14 TID 10152 Uninitialize:250 Destroying listener...                     
2020.5.12 6:46:14 TID 10152 Uninitialize:253 Destroying state notification object...                       
2020.5.12 6:46:14 TID 10152 FinalCleanUp:262 Entering final cleanup routine                                
CompletionContainerDestroy,92: Remove all 0 pending Echo.                                                  
1132196 rrwatchdog Process SysUpdate Not in running mode                                                   
1132830 rrwatchdog StartProc: UpdateProgress flag = UpdateProgress_Phase_0                 
1132832 rrwatchdog Notify Controller that SysUpdate encounter internal error 
`

I'll continue to test it.

JohnRev, 2020-05-12:
Thanks. I will test again, maybe I missed something.

JohnRev, 2020-05-12:
SysUpdate also needs patching:

```
sed -i 's/\x0B\x79\x03\xF0\xDF\x03\x53\x2B\x3E\xD0\x4C/\xCB\x78\x03\xF0\xDF\x03\x50\x2B\x3E\xD0\x4C/g' SysUpdate
```

Pattern:
```
Original Hex: 0B 79 03 F0 DF 03 53 2B 3E D0 4C
 Patched Hex: CB 78 03 F0 DF 03 50 2B 3E D0 4C
```

md5sum of patched SysUpdate for v2008 is `34439bd8c78a06e12c2e9d63ba1dafcf`

Tested it on v2008 and worked with flasher.py.

zvldz, 2020-05-12:
Works on firmware version 2008
Thanks.

I'll try other firmware.

zvldz, 2020-05-12:
But the patch doesn't work on gen1.
I'll try to see if it can be adapted.

JohnRev, 2020-05-12:
Only tried it on gen2 with firmware version 2008.
I can look into gen1. Which firmware introduced http OTA blocking?

zvldz, 2020-05-12:
I'm not sure, I'll ask the users.

zvldz, 2020-05-12:
On gen1, it seems to be working as it should.
There's no need to patch anything.
Thank you again.

JohnRev, 2020-05-12:
Thank you for providing this repo and the rooted FWs

zvldz, 2020-05-12:
Soon I will add the patch to the repository.

closed 2020-05-12 by zvldz