zvldz/vacuum #29: Robot not publishing its hostname on 2008

Poeschl, 2020-05-16:
Just noticed that the robot doesn't announces its hostname in the network on the 2008 firmware.

I can confirm that the script `custom_hostname.sh` sets the hostname correctly on the system. After that, I'm lost 😅
I guess that a system package is missing on the newer firmware.

zvldz, 2020-05-16:
How should the robot announce itself on the network ?

Poeschl, 2020-05-16:
I guess in the older firmware version something like avahi was installed to setup its own hostname via mDNS.

zvldz, 2020-05-16:
I'll think about how to fix it.

zvldz, 2020-05-31:
It's most likely a busybox limit.
But I don't know how to check the hostname via mDNS.
I tried it through avahi-resolve, but I don't get it (fw1896).

Hypfer, 2020-06-10:
You could try running wireshark to listen for mDNS broadcasts

zvldz, 2020-06-10:
It's not exactly an easy way.
But I will try.

dstolpmann, 2020-07-12:
I am running firmware version 2020 on a Roborock S5 and it does not seem to include the hostname in the DHCP request, which results in my DNS server not being updated and thus not being able to resolve the hostname.

zvldz, 2020-07-13:
Maybe it's fixed.
https://github.com/zvldz/vacuum/blob/master/custom-script/custom_vacuum.sh#L252

Poeschl, 2020-07-13:
I just re-flashed my robot with the current master and the custom hostname is set and reachable after some minutes. So this issue is fixed.

Thanks for the deep look into it @zvldz. I would say we wait for one more verification and close the issue afterwards.

Poeschl, 2020-07-14:
Hm, seems that after the daily reboot the hostname got lost. Just tested it and the name don't get resolved anymore.

zvldz, 2020-07-14:
It works for me even after reboot.

Poeschl, 2020-07-14:
Hm, rebooted the robot manually and it works fine. I will report back tomorrow again.

dstolpmann, 2020-07-14:
For me, the hostname is included in the DHCP requests, but only for a couple of minutes after boot. After that, the robot makes DHCP requests without the hostname.

Poeschl, 2020-07-15:
Worked for me until midnight with manual testing. This morning its gone again. Seems that something is overwriting the wifi_start script.

zvldz, 2020-07-15:
> Seems that something is overwriting the wifi_start script.

You can check the contents of the file to make sure.
My file is not overwritten by anything.

Poeschl, 2020-07-15:
I will make some research this evening

zvldz, 2020-07-15:
But it is possible that udhcpc does the next requests without  hostname

Poeschl, 2020-07-15:
So I checked the startup script and its untouched, sry for not checking first.
I also sniffed with Wireshark into the network and got the same two pairs of mdns requests before and after a fixing reboot.

First Variant:
```
Multicast Domain Name System (response)
    Transaction ID: 0xda27
    Flags: 0x8400 Standard query response, No error
    Questions: 0
    Answer RRs: 1
    Authority RRs: 0
    Additional RRs: 6
    Answers
        _services._dns-sd._udp.local: type PTR, class IN, _miio._udp.local
    Additional records
        _miio._udp.local: type PTR, class IN, roborock-vacuum-s5_miio117807017._miio._udp.local
        roborock-vacuum-s5_miio117807017._miio._udp.local: type TXT, class IN, cache flush
        roborock-vacuum-s5_miio117807017._miio._udp.local: type SRV, class IN, cache flush, priority 0, weight 0, port 54321, target roborock-vacuum-s5_miio117807017
        roborock-vacuum-s5_miio117807017: type A, class IN, cache flush, addr 192.168.0.33
        roborock-vacuum-s5_miio117807017: type A, class IN, cache flush, addr 192.168.0.33
        roborock-vacuum-s5_miio117807017: type NSEC, class IN, cache flush, next domain name roborock-vacuum-s5_miio117807017
    [Unsolicited: True]
```

Second variant:
```
Multicast Domain Name System (response)
    Transaction ID: 0x0000
    Flags: 0x8400 Standard query response, No error
    Questions: 0
    Answer RRs: 1
    Authority RRs: 0
    Additional RRs: 5
    Answers
        _miio._udp.local: type PTR, class IN, roborock-vacuum-s5_miio117807017._miio._udp.local
    Additional records
        roborock-vacuum-s5_miio117807017._miio._udp.local: type TXT, class IN, cache flush
        roborock-vacuum-s5_miio117807017._miio._udp.local: type SRV, class IN, cache flush, priority 0, weight 0, port 54321, target roborock-vacuum-s5_miio117807017
        roborock-vacuum-s5_miio117807017: type A, class IN, cache flush, addr 192.168.0.33
        roborock-vacuum-s5_miio117807017: type A, class IN, cache flush, addr 192.168.0.33
        roborock-vacuum-s5_miio117807017: type NSEC, class IN, cache flush, next domain name roborock-vacuum-s5_miio117807017
    [Unsolicited: True]
```
The requests are identical before and after reboot. Just after the reboot the hostname is working fine. Whats a little odd for me: There is no hostname included in the requests.

For comparison when I look  at the mdns request for my printer:
```
Multicast Domain Name System (query)
    Transaction ID: 0x0000
    Flags: 0x0000 Standard query
    Questions: 1
    Answer RRs: 0
    Authority RRs: 0
    Additional RRs: 0
    Queries
        MFC-J491DW.local: type A, class IN, "QM" question
    [Response In: 12065]
```

Poeschl, 2020-07-29:
Since the last comment I noticed that sometimes the robot is randomly available after a reboot and sometimes its only reachable via IP.

zvldz, 2020-07-29:
I'm out of ideas for now.
But I'm gonna try to watch.