zvldz/vacuum #30: No space left

mathiasrabe, 2020-05-22:
Hi everybody,

I would like to create a custom image with the following command:

```
sudo ./builder_vacuum.sh     --run-custom-script=ALL \
                                  --timezone=Europe/Berlin \
                                  --ntpserver=x.x.x.x \
                                  --public-key=../.ssh/id_rsa.pub \
                                  --enable-greeting \
                                  --disable-logs \
                                  --replace-adbd \
                                  --valetudo-path=./Valetudo \
                                  --replace-miio \
                                  --enable-dns-catcher \
                                  --enable-ramdisk \
                                  --fix-reset \
                                  --soundfile=./sounds/mysounds.pkg \
                                  -f ./firmware/v11_002008.fullos.fd043420-6ddb-4e54-bdb7-a8deec19f0fd.pkg
```

But I get error messages that the remaining space is not enough. I get messages like these:

```
sed: couldn't flush /some/folder/fw.cjOoh6/image/etc/sedATSz0N: Auf dem Gerät ist kein Speicherplatz mehr verfügbar
cat: Schreibfehler: Auf dem Gerät ist kein Speicherplatz mehr verfügbar 
tar:  ./etc/init/S09xdnsmasq: Funktion write fehlgeschlagen: Auf dem Gerät ist kein Speicherplatz mehr verfügbar
```

And so on. If I remove one command like --valetudo-path or --soundfile the file generation seems to work. My filesystem has about 6 GB left, so now I'm wondering what is the cause of these errors. Does that mean, that the root partition of the S50 is not large enough to handle all of my customs?

P.S. I'm working with Linux Mint in Virtual Box. Can this be a problem?

zvldz, 2020-05-22:
Try addштп
`--resize-root-fs=522240 \`
after 
`--soundfile=./sounds/mysounds.pkg \`

zvldz, 2020-05-22:
You need to update the builder.

You can also try to add
`--remove-extra-sounds \`

mathiasrabe, 2020-05-22:
Wow, you are fast 👍 

> Try addштп
> `--resize-root-fs=522240 \`
> after
> `--soundfile=./sounds/mysounds.pkg \`

This didn't help. The error messages remain. Maybe because the image will be resized at the end of the script?

> You need to update the builder.

I git cloned the repo about two hours ago.

> You can also try to add
> `--remove-extra-sounds \`

This helped. No errors are printed now. Thank you very much. Can you explain me what causes this issue?

zvldz, 2020-05-22:
Yes, already fixed. update and try again

zvldz, 2020-05-22:
The problem is that the 2008+ firmware has artificially reduced the root partition size

mathiasrabe, 2020-05-22:
Ok now `--resize-root-fs=522240 \` seems to work like expected.

Thanks for the fast help and the explanation ;)

closed 2020-05-22 by zvldz