zvldz/vacuum #35: Better firmware version check

bbarbisch, 2020-05-30:
custom_enable_local_ota.sh does apply binary patches to files specific to gen2 v2008 files. It might be a good idea to have a proper version check before applying those patches to avoid breaking the modified firmware.

A Version check could be like the following piece of code:
`
version=$(cat ${IMG_DIR}/etc/os-release | grep 'ROBOROCK_VERSION' | cut -d'_' -f3)
if [ "$version" -eq 2008 ]; then
    # do the patching
else
    # wrong version, don't apply patches
fi
`
This can (should?) be also added to other plugins specific to stripped down firmware versions (gen2 v2008ff) like the dropbear install script.

As I only own a gen2, I can't say if this also works on gen1 or other vacuums as I don't know if there is a /etc/os-release file with the ROBOROCK_VERSION entry on those vacuums.

zvldz, 2020-05-30:
When installing dropbear, the existing /etc/inittab file is checked.
Checking for a particular version(2008) is not correct. Since there is still version 2020 and there will be others.
As for patching, if the pattern is not found, the file remains unmodified.
Maybe then I'll add more tests, but I don't see the point yet.

bbarbisch, 2020-05-30:
if you want to check for a minimum version you can also to use the -ge operator
```
if [ "$version" -ge 2008 ]; then
...
```
IMHO it is a risky thing to do the patching before ensuring the correct firmware version(s) the patch should be applied to, because the search pattern might occur also in other (older) versions of the file (where changing might break its function). Which it does not at least for v1886 as the chosen search pattern seems to be long enough.

Also it might be possible to remove the md5 checksum checks after such a version check is in place (code cleanup).

The pattern you're looking for and the replacement might also change in future versions (dependent on further development, used compilers, optimization levels, obfuscation techniques,...)

zvldz, 2020-05-31:
I'll think about it later, now I'd like to finish the other tasks.

JohnRev, 2020-06-02:
hello @bbarbisch, original author of the ota patch here.
this is a good idea. for now i am 100% sure the patch does not affect older firmware versions.

very likely the pattern will need to be changed in the future, and we may need a version check then.

zvldz, 2020-06-02:
I'll do it later, now I want to finish the other tasks.
Maybe within a week.

zvldz, 2020-06-04:
done 24f673b094ef4697f8bac44a75f1e0f317a07cb7

closed 2020-06-07 by zvldz