zvldz/vacuum #40: Can not upgrade from 1898 to 2020

roth-m, 2020-06-22:
Hi,

First thank you for development of this project!
My cleaner has been running 1898 for some time now without any issue.

I tried to upgrade to 2020 (generating an image with the same options as with the 1898 firmware) using flasher (dustcloud), mirobo or Valetudo (url+md5) without success.
The new firmware is downloaded but after some minutes it reboots and says it kept the previous firmware as the new one is NOT OK.

Do I need to reset the cleaner to factory firmware to upgrade?
Is there a way to write the new firmware directly on the vacuum cleaner by manually (using bash) extracting the firmware and writing it to a partition (dd)?

Thanks!

zvldz, 2020-06-22:
Sometimes it happens that after firmware upgrade the vacuum returns to the previous firmware. 
The reason could be in watchdog's logs.
Installing firmware using dd is possible, but this is not the best option.

Resetting to factory settings can help.
But if there's no problem with the new firmware.

What options did you use when building the firmware?

roth-m, 2020-06-22:
Here is the command I use to build the firmware:

./builder_vacuum.sh     --run-custom-script=ALL \
                        --dnsserver=localiponmynetwork \
                        --ntpserver=localiponmynetwork \
                        --enable-appproxy-patcher \
                        --root-pass=myrootpwd \
                        --custom-user=myuser \
                        --custom-user-pass=mypass \
                        --enable-greeting \
                        --enable-rrlogd-patcher \
                        --replace-miio \
                        --enable-addon \
                        --enable-binding \
                        --enable-turn-off-ny \
                        --replace-adbd \
                        --valetudo-re-path=../Valetudo_RE \
                        --disable-firmware-updates \
                        --unprovisioned wpa2psk --ssid myssid --psk 'mypasswd' \
                        --set-host-cleaner \
                        -k $HOME/.ssh/id_ed25519.pub \
                        -f "$1"

(Arg1 / $1 is the stock image.)

I will take a look at the watchdog logs later and report here.

Thanks for your reply.

zvldz, 2020-06-22:
You have some misconceptions in your options.
> --enable-rrlogd-patcher

isn't required for Valetudo to work
>--disable-firmware-updates

Unnecessary for firmware with Valetudo
>-k $HOME/.ssh/id_ed25519.pub

2008+ firmware does not support Ed25519 by default
You should add --custom-dropbear

I'd also advise you to add:
--valetudo-re-nodeps
--enable-dns-catcher
--enable-local-ota
--resize-root-fs=522240
--enable-ramdisk (Optionally)
--disable-logs (Optionally)

But that doesn't explain why the firmware isn't installed :(

Watchdog logs must be watched immediately after the first reboot.

jansohn, 2020-06-22:
I had a similar problem. I got it working by using a pre-built image from https://vacuumz.info/download/gen2/.

I was also using `--unprovisioned` option in my custom build. Maybe that option isn't working correctly anymore?

zvldz, 2020-06-23:
Please update the builder.
And try again.

Maybe there was a problem with the new file custom_protect_ap.sh

roth-m, 2020-06-23:
Here is the watchdog log after flashing the 2020 custom firmware (with modified options you specified):
```

19538 rrwatchdog System is updating, phase-2 not start miio_client
19538 rrwatchdog sleep 0 s
20538 rrwatchdog ##################### 4. StartAllProcess done...
20900 rrwatchdog Process SysUpdate Not in running mode
20900 rrwatchdog Process miio_client Not in running mode
20900 rrwatchdog Process miio_client_helper_nomqtt.sh Not in running mode
20900 rrwatchdog ##################### 5. InitialMonitor done...
20901 rrwatchdog Boot Partition = 1
20902 rrwatchdog Successfully restore partition flag!
20902 rrwatchdog ##################### 6. Restore partition flag done...
20902 rrwatchdog ipc_com_rcv_thread started!

20903 rrwatchdog mem_monitor_thread started!

20903 rrwatchdog Offline partition = (2)
20904 rrwatchdog Offline partition flag = 2
20904 rrwatchdog Update Phase 1 install failed.
20905 rrwatchdog deleting pkginf file /mnt/data/.temp/Update.pkg.inf
20906 rrwatchdog Set return = 4
20906 rrwatchdog Send MSG_UPDATE_INSTALL_VERIFY (P1 failed)
20906 rrwatchdog watchdog send ipc msg = 17237880
_notify_daemon: Bad file descriptor
connect success!
21967 rrwatchdog Process: -1 miio_client disappeared??
22028 rrwatchdog miio_client crashed
40 0
22029 rrwatchdog rr_system: setsid miio_client -d /mnt/data/miio/ -l 2 >> /mnt/data/rockrobo/rrlog/miio.log 2>&1&
22036 rrwatchdog run shell script successfully. 
22036 rrwatchdog StartProc finished
22037 rrwatchdog Process: -1 miio_client_helper_nomqtt.sh disappeared??
22090 rrwatchdog miio_client_helper_nomqtt crashed
c0 0
22090 rrwatchdog rr_system: setsid miio_client_helper_nomqtt.sh >> /mnt/data/rockrobo/rrlog/miio.log 2>&1&
22096 rrwatchdog run shell script successfully. 
22096 rrwatchdog StartProc finished
```

I will rebuild the 1895 firmware (with options you specified) and retry to flash the 2020 on top.

roth-m, 2020-06-23:
From the very begining...
```

pChannelOpen,471: Failed to open pipe /tmp/WdRc for write, errno = 6
pChannelOpen,471: Failed to open pipe /tmp/WdAp for write, errno = 6
pChannelOpen,471: Failed to open pipe /tmp/WdMc for write, errno = 6
pChannelOpen,471: Failed to open pipe /tmp/WdUp for write, errno = 6
pChannelOpen,471: Failed to open pipe /tmp/WdWm for write, errno = 6
DllInit,1336: WatchDog loaded IPC.
DoInitialize,587: libroboipc v0.9 (Force Async). Initializing (Wd), pid = 725
pChannelOpen,529: Arrive event[Wd] 0x40ff0 created.
DllInit,1395: Arrive event[any] 0x41428 created.
7387 rrwatchdog mem size 230000/250000
7389 rrwatchdog UpdateProgress flag = 4
7389 rrwatchdog Update Global flag read = UpdateProgress_Phase_1_Post
7390 rrwatchdog UPDPKG_INFO_PATH from env: /mnt/data/.temp
7390 rrwatchdog UPDPKG_INFO_NAME from env: Update.pkg.inf
7390 rrwatchdog pkginf path: /mnt/data/.temp/Update.pkg.inf
7390 rrwatchdog version: v0.00.39
7390 rrwatchdog crash_time: 0 0
7390 rrwatchdog rr_system: pkill -9 miio
7445 rrwatchdog run shell script fail, script exit code:  1
7445 rrwatchdog run shell script fail, cmdline:  pkill -9 miio
7445 rrwatchdog rr_system: killall -9 wlanmgr
wlanmgr: no process found
7478 rrwatchdog run shell script fail, script exit code:  1
7478 rrwatchdog run shell script fail, cmdline:  killall -9 wlanmgr
7478 rrwatchdog rr_system: killall -9 AppProxy
AppProxy: no process found
7502 rrwatchdog run shell script fail, script exit code:  1
7502 rrwatchdog run shell script fail, cmdline:  killall -9 AppProxy
7502 rrwatchdog rr_system: killall -9 player
player: no process found
7525 rrwatchdog run shell script fail, script exit code:  1
7525 rrwatchdog run shell script fail, cmdline:  killall -9 player
7525 rrwatchdog rr_system: pkill -9 RoboController
7556 rrwatchdog run shell script fail, script exit code:  1
7556 rrwatchdog run shell script fail, cmdline:  pkill -9 RoboController
7556 rrwatchdog rr_system: killall -9 rrlogd
rrlogd: no process found
7579 rrwatchdog run shell script fail, script exit code:  1
7579 rrwatchdog run shell script fail, cmdline:  killall -9 rrlogd
7579 rrwatchdog Set Feed HW Watchdog interval to 16
7579 rrwatchdog WDIOC_SETFUNCCONFIG: 1
7583 rrwatchdog HW WDOG shutdown 9 1
7583 rrwatchdog HW WDOG shutdown close fd 9 0
7584 rrwatchdog watchdog hw_wdog_thread started!

7584 rrwatchdog Set Feed HW Watchdog interval to 16
7584 rrwatchdog WDIOC_SETFUNCCONFIG: 1
7590 rrwatchdog start to update mcu firmware /opt/rockrobo/firmware/img/RR_RUBY-S_MP_STM32.A3.G1_M3.3.0_RELEASE_20190801-101507_FULL.bin

7659 rrwatchdog update mcu firmware done 0
7683 rrwatchdog check_mcu_mode = 0
7683 rrwatchdog read boot reason = 15
7688 rrwatchdog merge_boot_reason ap reason = 15 mcu reason = 0 mcu err = 0 merge = 15
7693 rrwatchdog mcu time: Tue Jun 23 09:36:02 2020
 seconds:1592904962
7693 rrwatchdog ap  time: Tue Jun 23 09:36:03 2020
 seconds:1592897763 tz:-120
7693 rrwatchdog sync ap with mcu time: Tue Jun 23 09:36:02 2020

7693 rrwatchdog ##################### Normal sysmode = 0 ################
7694 rrwatchdog 
 +++ RRSEC WatchDoge+++ 

7697 rrwatchdog 
 +++ RRSEC digital sign verify pass from global path +++ 

environment works
[... 102 lines cut ...]
21338 rrwatchdog Send MSG_UPDATE_INSTALL_VERIFY (P1 failed)
21338 rrwatchdog watchdog send ipc msg = 17237880
_notify_daemon: Bad file descriptor
connect success!
22380 rrwatchdog Process: -1 miio_client disappeared??
22425 rrwatchdog miio_client crashed
40 0
22426 rrwatchdog rr_system: setsid miio_client -d /mnt/data/miio/ -l 2 >> /mnt/data/rockrobo/rrlog/miio.log 2>&1&
22432 rrwatchdog run shell script successfully. 
22432 rrwatchdog StartProc finished
22432 rrwatchdog Process: -1 miio_client_helper_nomqtt.sh disappeared??
22478 rrwatchdog miio_client_helper_nomqtt crashed
c0 0
22478 rrwatchdog rr_system: setsid miio_client_helper_nomqtt.sh >> /mnt/data/rockrobo/rrlog/miio.log 2>&1&
22485 rrwatchdog run shell script successfully. 
22485 rrwatchdog StartProc finished
22883 rrwatchdog Process SysUpdate Not in running mode
31270 rrwatchdog MEM Available:392760 at jiffies:22940 shmem:3140
wifi_start.sh: no process found
41310 rrwatchdog MEM Available:392808 at jiffies:38801 shmem:3184
51347 rrwatchdog MEM Available:392528 at jiffies:49239 shmem:3204
61386 rrwatchdog MEM Available:390792 at jiffies:56166 shmem:3236
Selected interface 'wlan0'

```

zvldz, 2020-06-23:
> I will rebuild the 1898 firmware (with options you specified) and retry to flash the 2020 on top.

Don't forget to update builder from github.

roth-m, 2020-06-23:
It has been updated before posting the logs.

zvldz, 2020-06-23:
There are several times a line
>System is updating, phase-2 not start miio_client

I haven't been able to reproduce the error yet.

zvldz, 2020-06-23:
I found an error in "unprovisioned mode" for 2008+ firmware.
But maybe that's not the solution yet.
I haven't had time to check it out yet.

https://github.com/zvldz/vacuum/commit/2f0f79bf7d61d8143f205d2c913da5a2b157e0e2

zvldz, 2020-06-23:
No, it's not working. It needs to be reworked.

zvldz, 2020-06-23:
fixed "unprovisioned mode" https://github.com/zvldz/vacuum/commit/6f87c031eab7cd55080d20df750d617ef1a7fbe6
But maybe you have another problem.

roth-m, 2020-06-24:
Flashed successfully!

It does not connect back to Wifi but it gives more hope ;)

Thanks

zvldz, 2020-06-24:
> It does not connect back to Wifi but it gives more hope ;)

It's weird, I was testing last night.
You can try to reset the wifi settings and connect to access point.

roth-m, 2020-06-24:
Reverted to 1898.

zvldz, 2020-09-03:
Closed #40

closed 2020-09-03 by zvldz