zvldz/vacuum #42: No Apt-Get in ready-built-firmwares > v1998?

willchr, 2020-07-02:
Hi,

I just installed an ready-built-firmware with firmware 2008 and valetudo 5.3 and after installing it turned out not to have apt-get installed on the robo.

Older firmwares doesn't have this issue.

Is there a reason for that? Can I install apt-get afterwards anyhow? dpkg isn't installed neither...  

Christian

zvldz, 2020-07-02:
Since version 2008, the firmware is no longer full ubuntu, but cutted os.
Somewhere I read that somebody tried to add apt themselves and it kind of worked.

willchr, 2020-07-02:
After searching a couple of hour I give up for the moment. If you find the article once again it would be nice if u could share it here...
In the past I installed python and alsasound in order to let the robo speak, so maybe there is another possibility to integrate these applications before flashing?

zvldz, 2020-07-03:
Valetudo Telegram group
https://t.me/joinchat/AR1z8xOGJQwkApTulyBx1w
[image]
[to-extract-from-root-with-tar-xhzvf.gz](https://github.com/zvldz/vacuum/files/4868427/to-extract-from-root-with-tar-xhzvf.gz)

What packages did you install exactly ?
There's a sox already installed to play the sound.

willchr, 2020-07-07:
Thank you very much!!

Having sox would be fine actually, but it doesn't seem being installed in my pkg with fw 2020 and valetudoRe9.6.
"man sox" doesn't work the same as "play" and "sox" don't work.

zvldz, 2020-07-07:
If you build the firmware yourself, you should add the option --enable-addon-sox.
The sox is installed in the ready-made firmware.

willchr, 2020-07-07:
hhm, strange, I didn't build it on my own.....
Maybe downloaded from another source...?.... Will download/install it again and see... thx for the moment!!!!

zvldz, 2020-09-03:
Closed #42

closed 2020-09-03 by zvldz