zvldz/vacuum #45: Reintroduce OOM adjustment on 2008+

bsdice, 2020-07-30:
When Roborock switched from Upstart to Busybox on 2008+ for Gen2, OOM adjustment for valetudo was lost. Reintroduce this feature for added operational safety. Implementation is through a sub shell, which adjusts its own OOM score and then execs the valetudo binary, replacing itself.

```sh
--- valetudo-daemon.sh.orig
+++ valetudo-daemon.sh
@@ -15,7 +15,12 @@
         echo "Running Valetudo"
         echo '|/bin/false' > /proc/sys/kernel/core_pattern
         if [ -f "/root/bin/busybox" ]; then
-            /root/bin/busybox ionice -c3 nice -n 19 /usr/local/bin/valetudo >> /var/log/upstart/valetudo.log 2>&1
+       (
+            # Make valetudo very likely to get killed when out of memory
+            echo 1000 > /proc/self/oom_score_adj
+            # Also run it with absolutely lowest CPU and I/O priority to not disturb anything critical on robot
+            exec /root/bin/busybox ionice -c3 nice -n19 /usr/local/bin/valetudo >> /var/log/upstart/valetudo.log 2>&1
+       )
         else
             nice -n 19 /usr/local/bin/valetudo >> /var/log/upstart/valetudo.log 2>&1
         fi
```

zvldz, 2020-07-30:
Thanks
added https://github.com/zvldz/vacuum/commit/46ea4810d10e8b5bc4957c15e23981ebf708bbc9

zvldz, 2020-08-03:
Closed #45

closed 2020-08-03 by zvldz