zvldz/vacuum #48: Firmware build installation

corentingosselin, 2020-08-26:
Hello back:)

I built my firmware on linux:

`prodigy@ubuntu:~/Desktop/vacuum-master$ sudo ./builder_vacuum.sh --firmware=v11_002020.pkg --run-custom-script=ALL --timezone=Europe/Berlin --enable-greeting --convert2eu --enable-addon-sox --enable-oucher --soundfile=french.pkg --public-key=id_rsa.pub 
Decrypt firmware
Unpack firmware
+ Generate SSH Host Keys if necessary
+ Replace ssh host keys
+ Disable SSH firewall rule
+ Unpacking addon-sox
+ Adding Greetings
+ Installing oucher (https://github.com/porech/roborock-oucher)
+ Installing custom sound files
++ Decrypt soundfile ..
++ Unpack soundfile ..
+ Removing obsolete authorized_keys from Xiaomi image
+ Add SSH authorized_keys
+ Replacing timezone
+ Prepare startup scripts
++ Added script to convert openssh host key to dropbear
++ Added script for init 'oucher'
++ Added script for change region to EU
+ Unpacking addon2
+ Setting hostname in dhcp request(fw 2008+)
+ Discard unused blocks
! If you install pigz, the firmware will be created faster (e.g. by (apt|dnf|zypper) install pigz)
Pack new firmware
Encrypt firmware
Copy firmware to ./output/vacuum_2020_2eu.pkg and creating checksums
5853dbfa024f85dffd1a372dd8c1f501  vacuum_2020_2eu.pkg
FINISHED
`

Then I uploaded it:
`MBP-de-prodigy:firmwarebuilder_0.3 prodigy$ python3 -m flasher --firmware vacuum_2020_2eu.pkg --token 3843426545364d4270554b6574365231
Flasher for Xiaomi Vacuum
Address is not set. Trying to discover.
Connecting to device 192.168.26.107...
Starting local http server...
Serving http server at 192.168.26.103:50633
Sending ota command with parameters: {"mode": "normal", "install": "1", "app_url": "http://192.168.26.103:50633/vacuum_2020_2eu.pkg", "file_md5": "5853dbfa024f85dffd1a372dd8c1f501", "proc": "dnld install"}
Ota started!
Progress: |--------------------------------------------------| 100.0% 
                                                                      
Firmware downloaded successfully.
Exiting.
`

The result:
`MBP-de-prodigy:firmwarebuilder_0.3 prodigy$ ssh root@192.168.26.107
Enter passphrase for key '/Users/prodigy/.ssh/id_rsa': 

BusyBox v1.24.1 (2019-11-15 15:22:50 CST) built-in shell (ash)

[root@rockrobo ~]# cd /
[root@rockrobo ]# ls
bin         boot        build.txt   dev         etc         home        lib         linuxrc     lost+found  mnt         opt         proc        root        run         sbin        sys         tmp         usr         var
[root@rockrobo ]# cat build.txt
build with dustbuilder (https://builder.dontvacuum.me)
Mon Aug 24 21:20:12 UTC 2020

[root@rockrobo ]# ls
bin         boot        build.txt   dev         etc         home        lib         linuxrc     lost+found  mnt         opt         proc        root        run         sbin        sys         tmp         usr         var`

the current build is an old build I tried with dustbuilder, this is not your vacuum build 
Also no files added, for example oucher.yml,oucher.conf, ect

Thanks for help

zvldz, 2020-08-27:
For some reason, the new firmware was not installed. Although the robot accepted it for installation.
Try resetting the robot to the factory settings and then reinstall the new firmware.

Or you can try to install the firmware again and look at the logs here /mnt/data/rockrobo/rrlog.
There may be a reason why the firmware is not installed.

corentingosselin, 2020-08-27:
Thank you so much, with factory reset, works like a charm !

corentingosselin, 2020-08-27:
Just small question, do you think is it possible to customize greeting sound ?

zvldz, 2020-08-28:
It seems that the file 
/opt/rockrobo/resources/sounds/power_on.wav 
or 
/opt/rockrobo/resources/sounds/start_greeting.wav
 is responsible for this

zvldz, 2020-09-03:
Closed #48

closed 2020-09-03 by zvldz