zvldz/vacuum #52: Feature Request: Add an option to set a remote syslog-server

pidator, 2020-11-27:
Could you implement a new parameter to configure/set a remote syslog-server (`IP:Port`) directly during image building?

For ubuntu-firmware it will be necessary to add the following line at the end of file `/etc/rsyslog.conf`:

`*.* @IP:Port`

For busybox firmware it will be necessary to replace line 3 of `/etc/init/S01logging`:

original: `SYSLOGD_ARGS=-n`
new: `SYSLOGD_ARGS="-n -R IP:Port -L -O /var/log/messages"`

Greetings

zvldz, 2020-11-27:
Done https://github.com/zvldz/vacuum/commit/3e0d03701f33b579535ea467435be0a9f02bbb36

But I didn't check on ubuntu fw.

pidator, 2020-11-27:
Thanks for the fast reaction.
Tested with my PR https://github.com/zvldz/vacuum/pull/53 and it's working on both firmwares!

pidator, 2020-11-29:
successful implemented - thanks :)

closed 2020-11-29 by pidator