zvldz/vacuum #56: DNS Catcher does not work

r7l, 2021-01-20:
Hello,

i have an issue with my S5 robot, not showing any map data and i was told it's a dns issue and was pointed to this builder. It's a great script. But somehow the DNS Catcher isn't working for me and i am still facing the map issues.

I've used this command to build the firmware:

```./builder_vacuum.sh --firmware=files/v11_002034.fullos.55915876-2190-407a-9fcb-f1e760d9b623.pkg --run-custom-script=ALL --replace-adbd --enable-appproxy-patcher --convert2eu --fix-reset --enable-local-ota --enable-history --ntpserver=192.168.X.X --enable-turn-off-ny --disable-logs --replace-miio --soundfile=files/my-sounds.pkg --public-key=files/my-key.pub --timezone=Europe/Berlin --valetudo-re-path=files/valetudo --hostname=MY-HOSTNAME --enable-addon --enable-dns-catcher --custom-dropbear```

According to the script it will add 2 lines to /etc/rc.local and i can see the lines in the file that ended up on the system. But if i run `iptables -L` it doesn't show those iptables settings and map data still doesn't work. I even tried to run those iptables commands manually but it doesn't change anything.

Can anyone help me with this?

zvldz, 2021-01-20:
> But if i run `iptables -L` it doesn't show those iptables settings

You have to run it like this
```
iptables -L -n -t nat
```
Try also looking at /var/log/upstart/valetudo.log

r7l, 2021-01-20:
I've did another image and flashed again. Not sure what went wrong the first time but seems to work for now. Thanks again for this awesome script.

closed 2021-01-20 by r7l