zvldz/vacuum #62: loosing data after reboot on _valetudo_2021_02_0.pkg

lukich48, 2021-02-22:
Hi! I created issue in valetudo repository but  maybe the problem is here
https://github.com/Hypfer/Valetudo/issues/706

Prebuild image vacuum_4028_valetudo_2021_02_0.pkg doesn't save config settings and it recreates clear /tmp/valetudo_config.json after each reboot

log after reboot:
```
[2021-02-20T15:02:19.602Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-02-20T15:02:19.610Z] [INFO] Autodetected RoborockV1ValetudoRobot
[2021-02-20T15:02:19.759Z] [INFO] Starting Valetudo 2021.02.0
[2021-02-20T15:02:19.760Z] [INFO] Configuration file: /tmp/valetudo_config.json
[2021-02-20T15:02:19.761Z] [INFO] Logfile: /tmp/valetudo.log
[2021-02-20T15:02:19.762Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. Xiaomi Mi SDJQR02RR (RoborockV1ValetudoRobot)
[2021-02-20T15:02:19.763Z] [INFO] JS Runtime Version v14.4.0
[2021-02-20T15:02:19.765Z] [INFO] Max Heap Size: 33.5 MiB
[2021-02-20T15:02:19.766Z] [INFO] Node Flags: --max-old-space-size=32
[2021-02-20T15:02:19.775Z] [INFO] DeviceId 82651588
[2021-02-20T15:02:19.776Z] [INFO] IP 127.0.0.1
[2021-02-20T15:02:19.778Z] [INFO] CloudSecret ogp36NPfP2wzhvHe
[2021-02-20T15:02:19.779Z] [INFO] LocalSecret f3Z7yeLTOiDEJSju
[2021-02-20T15:02:19.949Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-02-20T15:02:19.957Z] [INFO] Webserver running on port 80
[2021-02-20T15:02:19.964Z] [INFO] Map Upload Server running on port 8079
[2021-02-20T15:02:19.981Z] [WARN] Error during time sync: Error: queryA ECONNREFUSED pool.ntp.org
    at QueryReqWrap.onresolve [as oncomplete] (dns.js:203:19) {
  errno: undefined,
  code: 'ECONNREFUSED',
  syscall: 'queryA',
  hostname: 'pool.ntp.org'
}
[2021-02-20T15:02:27.873Z] [INFO] Cloud connected
[2021-02-20T15:03:19.004Z] [INFO] Successfully set the robot time via NTP to 2021-02-20T15:03:19.890Z
```

zvldz, 2021-02-22:
Yes, firmware contains old valetudo.conf.
There is no VALETUDO_CONFIG_PATH in it.
I'll recreate firmware images.

zvldz, 2021-02-22:
I recreated fw images.
Download and try install again.

lukich48, 2021-02-22:
Great! It works 👍

closed 2021-02-22 by lukich48

Rudi1952, 2021-03-08:
I have this issue also on valetudo_2021_02_0.
Can you give me a clue how this issue is solved?

zvldz, 2021-03-08:
Logs please or more details

Rudi1952, 2021-03-09:
Attached you will find the logfile

[2021-03-09T02:31:11.601Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-03-09T02:31:11.608Z] [INFO] Autodetected RoborockS5ValetudoRobot
[2021-03-09T02:31:11.755Z] [INFO] Starting Valetudo 2021.02.0
[2021-03-09T02:31:11.756Z] [INFO] Configuration file: /mnt/data/valetudo_config.json
[2021-03-09T02:31:11.756Z] [INFO] Logfile: /tmp/valetudo.log
[2021-03-09T02:31:11.757Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S5 (RoborockS5ValetudoRobot)
[2021-03-09T02:31:11.757Z] [INFO] JS Runtime Version v14.4.0
[2021-03-09T02:31:11.759Z] [INFO] Max Heap Size: 33.5 MiB
[2021-03-09T02:31:11.759Z] [INFO] Node Flags: --max-old-space-size=32
[2021-03-09T02:31:11.768Z] [INFO] DeviceId 261200022
[2021-03-09T02:31:11.768Z] [INFO] IP 127.0.0.1
[2021-03-09T02:31:11.769Z] [INFO] CloudSecret aY1VvW70MZYYkB5w
[2021-03-09T02:31:11.770Z] [INFO] LocalSecret XtaAnBW0RIlhs9dB
[2021-03-09T02:31:11.772Z] [INFO] Firmware Version: 3.5.8_002034
[2021-03-09T02:31:11.935Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-03-09T02:31:11.943Z] [INFO] Webserver running on port 80
[2021-03-09T02:31:11.949Z] [INFO] Map Upload Server running on port 8079
[2021-03-09T02:31:11.976Z] [WARN] Error during time sync: Error: queryA ECONNREFUSED pool.ntp.org
    at QueryReqWrap.onresolve [as oncomplete] (dns.js:203:19) {
  errno: undefined,
  code: 'ECONNREFUSED',
  syscall: 'queryA',
  hostname: 'pool.ntp.org'
}

zvldz, 2021-03-09:
Please show contents of /usr/local/bin/valetudo-daemon.sh

Rudi1952, 2021-03-09:
there is no file valetudo-deamon.sh in /usr/local/bin
there is only a binary file named 'valetudo'.

[root@rockrobo ]# ls -ls /usr/local/bin
 32189 -rwxr-xr-x    1 root     root      32961247 Feb 27 16:50 valetudo
[root@rockrobo ]#

zvldz, 2021-03-09:
Did you use ready made image or did you build it yourself?

zvldz, 2021-03-09:
You may not have updated the builder from the github.

Rudi1952, 2021-03-09:
I am not sure, i think i have used the ready made image

zvldz, 2021-03-09:
It's not like the images I created.
Maybe you have used another image.
You can download my image from https://vacuumz.info/download/gen2/ or use buidler from guthub.

Rudi1952, 2021-03-09:
I made an image with dustbuilder, with following:
* my own SSH-key
* Replace Xiaomi adbd with generic adbd (enables shell access via USB)
* Preinstall valetudo newest version
* Preinstall Nano texteditor, curl, wget, htop, hexdump
* Fixresets patch *new May 2020*
* S5 (ver 2034, 11/2020, stripped-Ubuntu)

Then i used the image v11_002034.pkg

zvldz, 2021-03-09:
dustbuilder uses its own builder, you need to contact with http://github.com/dgiese/.

Rudi1952, 2021-03-11:
Today i did a factory reset on my Roborock S50. After that i flashed 'vacuum_2020_valetudo_2021_02_0.pkg' on the S50.
I have done this with 'vacuumz' via my android.
I hoped to find a map after starting Valetudo, but unfortunately there is no map. I hope that you can help me.

Logs:
[2021-03-11T18:36:20.207Z] [INFO] Set Logfile to /tmp/valetudo.log
[2021-03-11T18:36:20.213Z] [INFO] Autodetected RoborockS5ValetudoRobot
[2021-03-11T18:36:20.363Z] [INFO] Starting Valetudo 2021.02.0
[2021-03-11T18:36:20.364Z] [INFO] Configuration file: /mnt/data/valetudo/valetudo_config.json
[2021-03-11T18:36:20.365Z] [INFO] Logfile: /tmp/valetudo.log
[2021-03-11T18:36:20.366Z] [INFO] Robot: Beijing Roborock Technology Co., Ltd. S5 (RoborockS5ValetudoRobot)
[2021-03-11T18:36:20.366Z] [INFO] JS Runtime Version v14.4.0
[2021-03-11T18:36:20.368Z] [INFO] Max Heap Size: 33.5 MiB
[2021-03-11T18:36:20.368Z] [INFO] Node Flags: --max-old-space-size=32
[2021-03-11T18:36:20.377Z] [INFO] DeviceId 261200022
[2021-03-11T18:36:20.378Z] [INFO] IP 127.0.0.1
[2021-03-11T18:36:20.379Z] [INFO] CloudSecret aY1VvW70MZYYkB5w
[2021-03-11T18:36:20.380Z] [INFO] LocalSecret Zz6OV5xvkPEPNuHl
[2021-03-11T18:36:20.382Z] [INFO] Firmware Version: 3.5.8_002034
[2021-03-11T18:36:20.556Z] [INFO] Dummycloud is spoofing 127.0.0.1:8053 on 127.0.0.1:8053
[2021-03-11T18:36:20.564Z] [INFO] Webserver running on port 80
[2021-03-11T18:36:20.571Z] [INFO] Map Upload Server running on port 8079
[2021-03-11T18:36:20.004Z] [INFO] Successfully set the robot time via NTP to 2021-03-11T18:36:20.890Z
[2021-03-11T18:37:06.056Z] [INFO] Cloud connected
[2021-03-11T18:37:06.075Z] [INFO] Unknown cloud message received: {"id":274208483,"prop":"ota_state","params":{"ota_state":"idle","ts":1615487765}}
[2021-03-11T18:37:08.086Z] [INFO] Unknown cloud message received: {"id":233551370,"prop":"ota_state","params":{"ota_state":"idle","ts":1615487828}}
[2021-03-11T18:38:10.759Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:39:11.643Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:40:12.679Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:41:13.720Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:42:14.760Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:43:15.780Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:44:16.804Z] [WARN] Failed to parse uploaded map
[2021-03-11T18:45:17.815Z] [WARN] Failed to parse uploaded map

zvldz, 2021-03-12:
Resetting to factory settings deletes all maps, you need to restart full cleaning to build new map